absalom-muc/MHI-AC-Ctrl #169: Why not using hardware SPI?

dzungpv, 2023-11-30:
This is not a bug, but why you are not using hardware SPI? I see code using **digitalWrite** and **digitalRead** to handle SPI communication.

hberntsen, 2023-12-01:
Mentioned [here](https://github.com/ginkage/MHI-AC-Ctrl-ESPHome/issues/43) before, I have an implementation with hardware SPI on an ESP32-C3 here: https://github.com/hberntsen/mhi-ac-ctrl-esp32-c3

glsf91, 2023-12-08:
I think we can close this one

closed 2023-12-08 by glsf91

oliv3r, 2025-01-16:
@absalom-muc coming back to this one for a bit, I'm trying to comprehend why this was not working a bit better.

I saw somewhere in the RTOS SDK documentation can't find it with a quick browse) that there is a limit to 20 bytes, surely, this is not what you ran into?

As for the

> cycle accurate synchronization between RX and TX

what did you mean? SPI by default is a fully _synchronized_ e.g. full-duplex protocol. E.g. when you clock in a byte, you will clock out a byte. For half-duplex situations, what is often done, is that simply zero's are sent out. E.g. as master, to request data from the slave, you just transmit 8 zero's, and you get 8 databytes back. Likewise, here as slave, we can only send data, when the master clocks us anything in (could be garbage or zero's of course).

This is because in effect, SPI is just a shift-register on each side.

So when you say they have to be clock cycle accurate, I'm not sure what you mean, and I wonder what you where struggling with.

oliv3r, 2025-01-16:
Yeah, I understand that it was a long time ago :)

So this should be handled by the simple fact that a proper SPI transaction is done, unless you need to modify bits during a transaction, e.g. bit 1 is received, which has an impact on bit 2 (or 3) (bit 1 is already pushed out :p)

If that where the case, you need to bit-bang SPI.

But indeed, the hardware SPI solution by hberendse suggests that it should be an issue of course. He has a 'problem' with that the ESP32 spi peripherial is only able to function when CS is being asserted, which is a different issue.