absalom-muc/MHI-AC-Ctrl #170: Can't send commands

wissenj, 2023-12-03:
I have installed version 2.8 with MQTT PubSubCLient v.2.7. D1 mini is soldered to the PCB Board which I manufactured by JLCPCB. This is mounted on the Unit SRK 50ZS-W. Via MQTT I can receive all data successfully. And I can send commands via MQTT, e.g. MHI-AC-Ctrl/set/Power which then get's confirmed by an MHI-AC-Ctrl/cmd_received o.k.

However, all send commands are not executed in any way. Seems they get lost on the way to the SRK unit.

I have checked the troubleshooting guide, but can't find the issue. My serial boot log:
```
Measure frequency for SCK, MOSI and MISO pin
SCK frequency=3200Hz (expected: >3000Hz) o.k.
MOSI frequency=560Hz (expected: <SCK frequency) o.k.
MISO frequency=0Hz (expected: ~0Hz) o.k.
OTA Ready
```
Also looks OK for me? Can someone give me a hint what to check next? Thank you very much!

absalom-muc, 2023-12-04:
It seems that the problem is related to the hardware.

- First I would check the PCB with a magnifying glass to ensure there is no bad solder joint (in most cases this is the problem).
- Please check if the Level Shifter is the right one. 
- Can you check if the MISO conncetion from PCB to the jack at the end of the cable is o.k.?
- Do you have electronic measurement equipment? E.g. an oscilloscope?
- Since it is simple to test you could change the ESP8266

btw: I appreciate the way how you described your problem - very clean and structured 👍

wissenj, 2023-12-07:
Hey absalom-muc,

Thank you for the nice help and the hints. My level shifter used had all necessary connections but wasn't fully bi-directional on all foru channels! And this was the reason. Maybe this can be added to the troubleshoot list or to the hardware description.

Thank you very much for the great project, looking forward now using it.

closed 2023-12-08 by glsf91