absalom-muc/MHI-AC-Ctrl #171: Add support for 10-17 degrees setting

tomm1ed, 2023-12-05:
See the 'MHI-AC-CTRL-Plus' project by 'Plusmore' on gitlab. He claims to be able to set the AC to 10-30 degrees from an ESP connected to the SPI port. Could this functionality maybe be added to this project as well?

glsf91, 2023-12-05:
You can use his solution if you want.
You can also archive this easily with your own automation. Just add some degrees to the Troom you use as an offset.

He added some stuff in the core files which don't belong there in my opinion. The core files are only to communicate with the AC. It is not the place for some additional logic.  We want this to keep this software as clean as possible.
So I'm not going to add this.

tomm1ed, 2023-12-05:
> You can use his solution if you want. You can also archive this easily with your own automation. Just add some degrees to the Troom you use as an offset.
> 
> He added some stuff in the core files which don't belong there in my opinion. The core files are only to communicate with the AC. It is not the place for some additional logic. We want this to keep this software as clean and stable as possible. I think absalom-muc will also agree to this. So I'm not going to add this.

The problem with using an automation is that you get dependancies while an integrated offset will keep on working even when HA/whatever else solution you use stops working. I do however understand what you are saying, I assumed he had found a way of telling the AC itself to achieve a specific temperature not fooling it by adjusting the tRoom on the fly.

Timple, 2023-12-05:
I concluded once that this could not be done from the outside: https://github.com/absalom-muc/MHI-AC-Ctrl/issues/118

Timple, 2023-12-05:
Is this request so much different than [the trick](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/SW-Configuration.md#enhance-resolution-of-tsetpoint-supporth) in this codebase to allow the 0.5 deg resolution?
They use the same idea to achieve an enhanced performance of the airconditioning. 

Of course we can use a fork, but why not enhance the main source?

tomm1ed, 2023-12-05:
> Is this request so much different than [the trick](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/SW-Configuration.md#enhance-resolution-of-tsetpoint-supporth) in this codebase to allow the [...]

Interesting point, one could make the case that it's quite the same.

glsf91, 2023-12-06:
> Is this request so much different than [the trick](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/SW-Configuration.md#enhance-resolution-of-tsetpoint-supporth) in this codebase to allow the 0.5 deg resolution?

Yes, that is different. That is added because a setpoint of x.5 degrees was not working the right way. 
The setpoint with .5 degrees will be used by a lot more people.

glsf91, 2023-12-17:
I think we can close this one

closed 2023-12-17 by glsf91

yasinCelik1, 2024-01-11:
> The setpoint with .5 degrees will be used by a lot more people.

I actually have the same request. A lot of people living in colder climates wan't to use setpoints below 18 degrees. Right now I use the ECO function all the time with the RC. It creates an offset of -2.5 degrees after two hours of runtime. It can only be activated by RC and will be turned off after changing fanmode. Problem is that ECO is also limiting compressor speed, and a setpoint below 15.5 degrees is not possible. Also there is no need for an external temperature sensor with an SRF-unit for me since it is accurate.

Is it something that could be optionally turned on in the code?

glsf91, 2024-01-11:
> Also there is no need for an external temperature sensor with an SRF-unit for me since it is accurate.

It will add a lot of complexity with an internal temperature sensor to this project which I don't like. Also it should not be located in de core files. The core files are for the communication to the AC, not the place for logic (messing around with the internal temp sensor). Outside the core files it will also be more complicated.

If you want use this with an internal temp sensor, use the project mentioned in the first post.