absalom-muc/MHI-AC-Ctrl #172: Wrong runtime counters

KoelnSolar, 2023-12-13:
Hi there,

great job ! 

Since yesterday my SRK35-ZSX-WB with SRC35-ZSX-W is running with Wemos-mini. I just added an LED to reduce power of 12,5V(measured) to a voltage beneath 12V(3.3 voltage regulator has max. voltage 12V). Level shifting MOSI/SCLK is done by voltage dividers 825/2200. USE_EXTENDED_FRAME_SIZE is enabled. Version MHI-AC-Ctrl is 2.8.

OpData_TOTAL-IU-RUN didn't change so far and still shows 100(value seems to be 1)
OpData_TOTAL-COMP-RUN was 100(value seems to be 1) and changed now the first time to 200(last power off was yesterday 4 p.m.)

further issues:
OpData_CT seems to be calculated wrong(at least 0.5 higher than measured consumption). Maybe it is an issue that no wave shift Phi is given ?

OpData_OUTDOOR is nearly 1 degree lower than temperature measured with DS18B20

setting fan speed doesn't work for 2 and 3. setting 4 results in level 7. Auto and level 5,6,7 as settings are missing(checked against program code). 

I'm missing settings for "Hi/Lo"(boost/eco mode), silent mode....

Any idea of calculating thermal energy and COP ?

I'm using FHEM as MQTT Server and have a lot of possibilities to compare data against measurements, program things...... Don't hesitate to contact me if you need further informations or assistance.

Regards, Markus

glsf91, 2023-12-13:
OpData_CT: 
We cannot change this, this is wat the AC provides. The voltage is not always 230V, so multiplying always with 230V is not right. Better to multiply with voltage from P1 meter measured (on the right phase).

OpData_OUTDOOR:
We cannot change this, this is wat the AC measures and provides.

Fan speed:
level 5,6,7 don't exists. Only 1-4 and Auto like on the remote control. The levels 1-8 are for OpDate/IU-Fanspeed and are different from topic Fan. 
So if you set Fan to 3, the IU-Fanspeed will normally become 6. BUT the AC can change this to lower values.
So if you set Fan, look into the topic Fan and you will see the same value (and means it is working). Same for Auto,

"Hi/Lo"(boost/eco mode), silent mode...." is not possible.

KoelnSolar, 2023-12-13:
Thanks for quick reply.

> The voltage is not always 230V, so multiplying always with 230V is not right. Better to multiply with voltage from P1 meter measured (on the right phase).

Thats true, but multiplying 230V is OK to get nearly the power. 25% difference is caused by wrong CT, maybe missing wave shift.

> We cannot change this, this is wat the AC provides. 

The value is not provided by AC. It is calculated `dtostrf(value * 14 / 51.0f, 0, 2, strtmp)`

> OpData_OUTDOOR:
We cannot change this, this is wat the AC measures and provides.

Is calculated too. `dtostrf((value - 94) * 0.25f, 0, 2, strtmp)`

> "Hi/Lo"(boost/eco mode), silent mode...." is not possible.

Currently in software and not yet evaluated, but should be supported by SPI, I think.

glsf91, 2023-12-13:
Of course they are calculated but value is coming from AC en calculations are right. And yes this can differ from own measured values.

> Currently in software and not yet evaluated, but should be supported by SPI, I think.

Is not supported by SPI.

KoelnSolar, 2023-12-13:
How do you know the right formulas for calculations ?

At lunch time I made a test to check directly power after each change of AC-CT. As you can see the differences are not that big. Maybe just time shifted. 
But in low power mode CT didn't change since 4:24 p.m . Real power is 20% less. As you can see, I checked against power meter and grid.
I'm wondering that I could see only this 1.1. And there was only one lower value 0.82. Is it similar with your ACs ?
It seems that CT is only a kind of calculation and not really measured.

[AC power.pdf](https://github.com/absalom-muc/MHI-AC-Ctrl/files/13664164/AC.power.pdf)

glsf91, 2023-12-13:
> How do you know the right formulas for calculations ?

Because in the past they are compared with other devices connected with the AC and showing the values.

> Is it similar with your ACs ?

I don't measure the current with a power meter so cannot compare with CT.
And I do not calculate power usage with CT so can also not compare with power usage from powermeter.
It will be always preferable to use the power measured from your powermeter if you have one. It will be more accurate.

glsf91, 2023-12-21:
I think we can close this one

closed 2023-12-21 by glsf91

KoelnSolar, 2023-12-21:
Why did you close the issue ?
Was there any answer regarding the runtime counters ?

glsf91, 2023-12-21:
Because there was no activity anymore.
> Was there any answer regarding the runtime counters ?

It is in the last answer. Calculations are based (in the past) on a MHI device which was also displaying these values.