absalom-muc/MHI-AC-Ctrl #174: Enhanced resolution 'true' means less accurate Troom

th3sniper86, 2023-12-22:
I've set the following parameters in support.h:
#define TROOM_FILTER_LIMIT 0                        .
#define ENHANCED_RESOLUTION true

I'm using a thermostat that is very precise (0.1 degree). Every temperature update is send to the AC using MQTT.
I've noticed that when the mqtt temperature is recieved, the AC translates this to a different Troom:

Tsetpoint 21.5 (MQTT) = Tsetpoint 22 (AC):
MQTT actual temp. = AC Troom (difference)
21	 = 21.5	(+0.5)
20.9 = 21.25 (+0.35)
20.8 = 21.25 (+0.45)
20.7 = 21 (+0.3)
20.6 = 21 (+0.4)
20.5 = 21 (+0.5)
20.4 = 20.75 (+0.35)
20.3 = 20.75 (+0.45)
20.2 = 20.5 (+0.3)
20.1 = 20.5 (+0.4)
20.0 = 20.5 (+0.5)
19.9 = 20.25 (+0.35) 
19.8 = 20.25 (+0.45)
19.7 = 20 (+0.3)
19.6 = 20 (+0.4)
19.5 = 20 (+0.5)
19.4 = 19.75 (+0.35)
19.3 = 19.75 (+0.45)
19.2 = 19.5 (+0.3)
19.1 = 19.5 (+0.4)
19.0 = 19.5 (+0.5)
18.9 = 19.25 (+0.35)

Tsetpoint 21 (MQTT/AC):
20.6 = 20.5 (-0.1)
20.5 = 20.5 (0)
20.4 = 20.25 (-0,15)
20.3 = 20.25 (-0,05)
20.2 = 20 (-0.2)
20.1 = 20 (-0.1)
20.0 = 20 (0)
19.9 = 19.75 (-0,15)

This means the accuracy is somewhere between 0.3 and 0.5 degrees when using 'ENHANCED_RESOLUTION' and 0.0 and 0.2 when NOT using 'ENHANCED_RESOLUTION.  This would result in a more fluctuating room temperature if you use enhanced resolution (setpoint x.5). Is this expected behavior?

[image]

Thanks!

glsf91, 2023-12-22:
The setpoint used by the AC can only be in whole degrees, so for example 20 or 21 degrees. No 20.5 degrees will be used by the AC. So if you set a setpoint of 20.5 degrees, the AC will use a setpoint of 21 degrees. That is 0.5 degrees to high.

To compensate for this you can use the ENHANCED_RESOLUTION. When you use ENHANCED_RESOLUTION the Troom will be increased every time with 0.5 degrees. 
So if you use 20.5 degrees as a setpoint and Troom is 20.0 degrees the AC while use a setpoint of 21 degrees and a Troom of 20.5. It will heat for the 0.5 degrees difference.

And why do you see "20.3 = 20.75 (+0.45)":
The Troom send to the AC or received from the AC has a resolution of .25 degrees. That means if you send 20.3 by MQTT there will be (with setpoint 20.5) an offset of 0.5 added (see above). That will make 20.8. But the AC only accepts in steps of 0.25 degrees. So the AC will receive 20.75 degrees. As you discovered sending 20.3 or 20.4 makes no difference for the AC.
Basically: using a temperature sensor with a higher resolution then 0.25 makes not much sense.

TROOM_FILTER_LIMIT : Defines from which Troom delta value a new Troom value is published.
So assume the AC is sending out a Troom of 20.0 (which was published) and the AC is sending next time a Troom of 20.25 . Then with a TROOM_FILTER_LIMIT  of 0 the new value of 20.25 will be published. But if you use a TROOM_FILTER_LIMIT of 0.25 it will not be published. The next value of 20.5 will be published in this case (remember it can only be in 0.25 increments).

So if you want really want a resolution of .5 degrees in setpoint, use ENHANCED_RESOLUTION.
If you want to see every increment of the Troom (from the AC) use TROOM_FILTER_LIMIT 0.0

th3sniper86, 2023-12-22:
OK I see it now. If I indeed deduct 0,5 degrees from Troom when using Enhanced resolution, I get the same accuracy between what the AC is showing and what my thermostat shows. 
Thanks!

closed 2023-12-22 by th3sniper86