absalom-muc/MHI-AC-Ctrl #177: Update from 2.02 to 2.8 doesn't work

coyote1, 2024-01-10:
Hi, have been using your project since 2020 without any problems with currently SW version 2.02.
Now I wanted to upgrade the software version to the latest version 2.8. 
So I flashed a new Wemos D1 Mini and put it on the PCB. Unfortunately it doesn't work and I don't know why.
Is the SW V2.8 compatible with the "old" PCB?

According to the log, it looks to me like the ESP keeps rebooting, doesn't it?
I have flashed the Wemos with 80MHz and also with 160MHz, but nothing has changed.
Maybe you can help me, otherwise I'll just stick with the "old" version, it works without any problems :-)

Name of my AC: SRK50-ZM-S
Used MHI-AC-Ctrl HW: PCB from Dec 2019 (no HW Version Number on PCB)
Used MHI-AC-Ctrl SW: 2.02

Log:

```
17:31:57.024 -> Starting MHI-AC-Ctrl v2.8
17:31:57.024 -> CPU frequency[Hz]=160000000
17:31:57.024 -> ESP.getCoreVersion()=3.1.2
17:31:57.024 -> ESP.getSdkVersion()=2.2.2-dev(38a443e)
17:31:57.316 -> ESP.checkFlashCRC()=1
17:31:57.625 -> Measure frequency for SCK, MOSI and MISO pin
17:31:58.630 -> SCK frequency=4299Hz (expected: >3000Hz) o.k.
17:31:58.630 -> MOSI frequency=590Hz (expected: <SCK frequency) o.k.
17:31:58.630 -> MISO frequency=0Hz (expected: ~0Hz) o.k.
17:31:58.630 -> OTA Ready
17:31:58.630 -> setupWiFi: Start async scanNetworks
17:31:58.630 -> WIFI_CONNECT_SCANNING
17:31:58.697 -> mhi_ac_ctrl_core.loop error: -2
17:31:58.768 -> mhi_ac_ctrl_core.loop error: -2
17:31:58.844 -> mhi_ac_ctrl_core.loop error: -2
17:31:58.915 -> mhi_ac_ctrl_core.loop error: -2
17:31:58.992 -> mhi_ac_ctrl_core.loop error: -2
17:31:59.062 -> mhi_ac_ctrl_core.loop error: -2
17:31:59.134 -> mhi_ac_ctrl_core.loop error: -2
17:31:59.207 -> mhi_ac_ctrl_core.loop error: -2
17:31:59.277 -> mhi_ac_ctrl_core.loop error: -2
17:31:59.380 -> mhi_ac_ctrl_core.loop error: -2
17:31:59.454 -> mhi_ac_ctrl_core.loop error: -2
17:31:59.525 -> mhi_ac_ctrl_core.loop error: -4
17:31:59.600 -> mhi_ac_ctrl_core.loop error: -4
17:31:59.671 -> mhi_ac_ctrl_core.loop error: -2
17:31:59.744 -> mhi_ac_ctrl_core.loop error: -2
17:31:59.819 -> mhi_ac_ctrl_core.loop error: -2
17:31:59.930 -> mhi_ac_ctrl_core.loop error: -2
17:31:59.967 -> mhi_ac_ctrl_core.loop error: -2
17:32:00.075 -> mhi_ac_ctrl_core.loop error: -2
17:32:00.148 -> mhi_ac_ctrl_core.loop error: -2
17:32:00.194 -> mhi_ac_ctrl_core.loop error: -2
17:32:00.304 -> mhi_ac_ctrl_core.loop error: -2
17:32:00.373 -> mhi_ac_ctrl_core.loop error: -2
17:32:00.449 -> mhi_ac_ctrl_core.loop error: -2
17:32:00.517 -> mhi_ac_ctrl_core.loop error: -4
17:32:00.593 -> mhi_ac_ctrl_core.loop error: -4
17:32:00.668 -> mhi_ac_ctrl_core.loop error: -2
17:32:00.741 -> mhi_ac_ctrl_core.loop error: -2
17:32:00.816 -> mhi_ac_ctrl_core.loop error: -2
17:32:00.883 -> mhi_ac_ctrl_core.loop error: -2
17:32:00.988 -> mhi_ac_ctrl_core.loop error: -2
17:32:01.062 -> mhi_ac_ctrl_core.loop error: -2
17:32:01.138 -> mhi_ac_ctrl_core.loop error: -2
17:32:01.209 -> mhi_ac_ctrl_core.loop error: -2
17:32:01.281 -> mhi_ac_ctrl_core.loop error: -2
17:32:01.349 -> mhi_ac_ctrl_core.loop error: -2
17:32:01.422 -> mhi_ac_ctrl_core.loop error: -2
17:32:01.496 -> mhi_ac_ctrl_core.loop error: -2
17:32:01.567 -> mhi_ac_ctrl_core.loop error: -4
17:32:01.675 -> mhi_ac_ctrl_core.loop error: -4
17:32:01.744 -> mhandleWiFiScanResult(): 10 access points available
17:32:01.744 ->  1              TP-Link_84FE  1 -93dBm 98:DA:C4:68:3E:31 secured 08
[... 98 lines cut ...]
17:32:05.428 -> ⸮⸮n⸮r⸮⸮n|⸮l⸮l`bbrl⸮nb⸮nl`⸮rl⸮l⸮⸮
17:32:05.537 -> Starting MHI-AC-Ctrl v2.8
17:32:05.537 -> CPU frequency[Hz]=160000000
17:32:05.537 -> ESP.getCoreVersion()=3.1.2
17:32:05.537 -> ESP.getSdkVersion()=2.2.2-dev(38a443e)
17:32:05.824 -> ESP.checkFlashCRC()=1
17:32:06.147 -> Measure frequency for SCK, MOSI and MISO pin
17:32:07.127 -> SCK frequency=4280Hz (expected: >3000Hz) o.k.
17:32:07.127 -> MOSI frequency=587Hz (expected: <SCK frequency) o.k.
17:32:07.127 -> MISO frequency=0Hz (expected: ~0Hz) o.k.
17:32:07.127 -> OTA Ready
17:32:07.127 -> setupWiFi: Start async scanNetworks
17:32:07.160 -> WIFI_CONNECT_SCANNING
17:32:07.194 -> mhi_ac_ctrl_core.loop error: -2
17:32:07.268 -> mhi_ac_ctrl_core.loop error: -2
17:32:07.340 -> mhi_ac_ctrl_core.loop error: -2
17:32:07.416 -> mhi_ac_ctrl_core.loop error: -2
17:32:07.487 -> mhi_ac_ctrl_core.loop error: -2
17:32:07.561 -> mhi_ac_ctrl_core.loop error: -2
17:32:07.633 -> mhi_ac_ctrl_core.loop error: -2
17:32:07.743 -> mhi_ac_ctrl_core.loop error: -4
17:32:07.817 -> mhi_ac_ctrl_core.loop error: -4
17:32:07.889 -> mhi_ac_ctrl_core.loop error: -2
17:32:07.964 -> mhi_ac_ctrl_core.loop error: -2
```

Henk50V, 2024-01-10:
Have you seen this in the readme? I had the same issue
⚠️[MQTT client library]
..... don't use v2.8.0,  use v2.7.0

coyote1, 2024-01-10:
Sure. I know that, because I use this Project since 2020.
But I will check the PubSub client version and flash it again.

glsf91, 2024-01-10:
Indeed you esp is crashing a few times. 
A lot have changed in between :-)
If it is not PubSub client version you can also try:
- without framesize 33 if you have enabled this in support.h 
- try an earlier version

Henk50V, 2024-01-10:
I see in your log it is near wifi connection, in my case,  after months of stable working, without network changes, connected goes nearly all the time to 0. Didn't  connect to serial to debug. But you can give it a try to change parameter:
#define WiFI_SEARCHStrongestAP false
For me, this solved the problem.

coyote1, 2024-01-11:
No luck.
I tried all options: Without frame size, StrongestAP false, both frequencies.
MQTT connection is there, I can set Power ON, but it doesn't work.

I also tried ESPHome Version, all Datapoints are there, but I can't control MHI.
Here are serial  log and pictures from ioBroker.
Maybe hardware layout is different?!

```
17:26:33.987 -> Starting MHI-AC-Ctrl v2.8
17:26:33.987 -> CPU frequency[Hz]=80000000
17:26:33.987 -> ESP.getCoreVersion()=3.1.2
17:26:33.987 -> ESP.getSdkVersion()=2.2.2-dev(38a443e)
17:26:34.527 -> ESP.checkFlashCRC()=1
17:26:34.820 -> Measure frequency for SCK, MOSI and MISO pin
17:26:35.824 -> SCK frequency=4320Hz (expected: >3000Hz) o.k.
17:26:35.824 -> MOSI frequency=577Hz (expected: <SCK frequency) o.k.
17:26:35.824 -> MISO frequency=0Hz (expected: ~0Hz) o.k.
17:26:35.861 -> OTA Ready
17:26:35.861 -> setupWiFi: Start async scanNetworks
17:26:35.861 -> WIFI_CONNECT_SCANNING
17:26:35.899 -> mhi_ac_ctrl_core.loop error: -2
17:26:35.969 -> mhi_ac_ctrl_core.loop error: -2
17:26:36.040 -> mhi_ac_ctrl_core.loop error: -2
17:26:46.112 -> mhi_ac_ctrl_core.loop error: -2
17:26:46.183 -> mhi_ac_ctrl_core.loop error: -2
17:26:46.259 -> mhi_ac_ctrl_core.loop error: -2
17:26:46.333 -> mhi_ac_ctrl_core.loop error: -2
17:26:46.333 ->  connected to CoyoteIoT, IP address: 192.168.59.110 (-84dBm)
17:26:46.333 -> WIFI_CONNECT_OK
17:26:46.406 -> mhi_ac_ctrl_core.loop error: -2
17:26:46.406 -> MQTTreconnect(): Attempting MQTT connection (MQTTclient.state=-1), WiFi.status()=3 ...
17:26:47.590 ->  connected
17:26:47.590 -> MQTTclient.connected=1
17:26:47.590 -> status=64 topic=connected payload=1
17:26:47.590 -> status=64 topic=Version payload=2.8
17:26:47.624 -> status=64 topic=RSSI payload=-84
17:26:47.624 -> status=64 topic=WIFI_LOST payload=0
17:26:47.624 -> status=64 topic=MQTT_LOST payload=0
17:26:47.624 -> status=64 topic=WIFI_BSSID payload=4E:D9:E7:FA:DB:16
17:26:47.624 -> MQTTreconnect(): 10 access points available
17:26:47.624 -> status=64 topic=fSCK payload=4320
17:26:47.624 -> status=64 topic=fMOSI payload=577
17:26:47.624 -> status=64 topic=fMISO payload=0
17:26:47.700 -> mhi_ac_ctrl_core.loop error: -2
17:26:47.773 -> mhi_ac_ctrl_core.loop error: -2
17:26:47.848 -> mhi_ac_ctrl_core.loop error: -2
17:26:47.925 -> mhi_ac_ctrl_core.loop error: -2
17:26:48.001 -> mhi_ac_ctrl_core.loop error: -4
17:29:18.030 -> mhi_ac_ctrl_core.loop error: -2
17:29:18.105 -> mhi_ac_ctrl_core.loop error: -2
17:29:18.105 -> MQTT_subscribe_callback, topic=MHI_Wintergarten2/set/Power payload=On payload_length=2
17:29:18.105 -> status=64 topic=cmd_received payload=o.k.
17:29:18.177 -> mhi_ac_ctrl_core.loop error: -2
17:29:18.283 -> mhi_ac_ctrl_core.loop error: -4
17:29:18.353 -> mhi_ac_ctrl_core.loop error: -4
17:29:18.428 -> mhi_ac_ctrl_core.loop error: -2
17:31:24.095 -> mhi_ac_ctrl_core.loop error: -2
17:31:24.170 -> mhi_ac_ctrl_core.loop error: -2
17:31:24.242 -> mhi_ac_ctrl_core.loop error: -2
17:31:24.242 -> MQTT_subscribe_callback, topic=MHI_Wintergarten2/set/Power payload=Off payload_length=3
17:31:24.242 -> status=64 topic=cmd_received payload=o.k.
17:31:24.314 -> mhi_ac_ctrl_core.loop error: -2
17:31:24.388 -> mhi_ac_ctrl_core.loop error: -2
17:31:24.463 -> mhi_ac_ctrl_core.loop error: -2
17:31:24.534 -> mhi_ac_ctrl_core.loop error: -2
```

[image]

glsf91, 2024-01-11:
It is not crashing anymore :-)
Looks like there is no good communication with the AC because after startup it should publish almost all topics in about 20-40 sec. Or did you remove some of the logging?
Give version 2.6 a try because between 2.6 and 2.8 communication with AC was also changed a little bit.

coyote1, 2024-01-11:
Yes, I have removed some loop errors in the log so that it is shorter.
I will test the 2.6, maybe it will work.
It's no problem, the 2.02 works without any problems. Just thought I'd upgrade to the latest version.
It's strange that the ESPHome version doesn't work either.
I can also send commands via iobroker and it says command ok

glsf91, 2024-01-12:
> It's strange that the ESPHome version doesn't work either.

That is not strange because it uses the same core files (which are doing the communication with the AC).

coyote1, 2024-01-13:
@glsf91  v2.6 works fine

glsf91, 2024-01-13:
> @glsf91 v2.6 works fine
That is nice.
Do you still have a lot of mhi_ac_ctrl_core.loop errors with v2.6 ?

coyote1, 2024-01-13:
I would say it is limited :-)

```
23:06:04.249 -> setupWiFi2:12 access points available
23:06:04.249 -> current BSSID: 22:E8:29:A7:77:0D, strongest BSSID: 4E:D9:E7:FA:DB:16
23:06:06.911 -> status=135 topic=THI-R1 payload=15
23:06:08.788 -> mhi_ac_ctrl_core.loop error: -2
23:06:42.821 -> mhi_ac_ctrl_core.loop error: -2
23:07:05.947 -> mhi_ac_ctrl_core.loop error: -2
23:07:06.607 -> mhi_ac_ctrl_core.loop error: -2
23:07:09.203 -> status=135 topic=THI-R1 payload=14
23:07:33.816 -> mhi_ac_ctrl_core.loop error: -2
23:07:40.148 -> status=135 topic=THI-R1 payload=15
23:07:51.015 -> mhi_ac_ctrl_core.loop error: -2
23:07:51.840 -> mhi_ac_ctrl_core.loop error: -2
23:07:54.890 -> mhi_ac_ctrl_core.loop error: -2
23:08:11.312 -> status=135 topic=THI-R1 payload=14
```

So I will probably stick with v2.6.
Let's see if I can get the ESPHome version to work with the v2.6 core files.

glsf91, 2024-01-14:
I think the communication with you AC is a little bit critical for some reason. I have almost never a mhi_ac_ctrl_core.loop error.
I also think the change in v2.7R3 (change sending MISO with faster refresh of data every 20s) will be the cause of the problem for you.

If you want you can first try v2.7R2 (https://github.com/absalom-muc/MHI-AC-Ctrl/tree/7814a9d9942fcb288c5473e5fe2ec8b4d8efc8da) which should also work.
And I expect if you try v2.7R3 (https://github.com/absalom-muc/MHI-AC-Ctrl/tree/e9bccf4e47f0a615186739b23c69d6fbadbe3cea) it won't work anymore.

If this is the case we can also try v2.8 again with a very little change to see if this solves the problem. If you want of course.

coyote1, 2024-01-14:
Don't ask me why, but now it's suddenly working.

2.7R2 flashed -> works
2.7R3 flashed -> works
2.8 flashed, works now too

The only thing that was different now was that I restarted the laptop and flashed the ESP when it was plugged directly onto the board and connected to the air conditioning.

Perhaps I will now test the ESPHome version.

Log:

```
14:22:43.752 -> status=137 topic=THI-R3 payload=14
14:22:44.662 -> status=134 topic=IU-FANSPEED payload=0
14:22:45.573 -> status=139 topic=TOTAL-IU-RUN payload=200
14:22:46.508 -> status=132 topic=OUTDOOR payload=6.00
14:22:47.433 -> status=133 topic=THO-R1 payload=-4
14:22:48.352 -> status=141 topic=COMP payload=0.00
14:22:49.300 -> status=143 topic=TD payload=<=30
14:22:50.215 -> status=142 topic=CT payload=0.00
14:22:50.994 -> WIFI_CONNECT_TIMEOUT
14:22:51.098 -> setupWiFi: Start async scanNetworks
14:22:51.098 -> WIFI_CONNECT_SCANNING
14:22:52.836 -> status=144 topic=TDSH payload=1
14:22:53.826 -> handleWiFiScanResult(): 10 access points available
14:22:53.826 ->  1              TP-Link_84FE  1 -89dBm 98:DA:C4:68:3E:31 secured 08
14:22:53.826 ->  2                    Coyote  1 -77dBm 18:E8:29:A7:77:0D secured 04
14:22:53.826 ->  3                CoyoteGast  1 -74dBm 1E:E8:29:A7:77:0D secured 04
14:22:53.826 ->  4                 CoyoteIoT  1 -75dBm 22:E8:29:A7:77:0D secured 04
14:22:53.826 ->  5                CoyoteGast  1 -89dBm 7E:8A:20:2D:D9:4C secured 04
14:22:53.826 ->  6                 CoyoteIoT  1 -93dBm 82:8A:20:2D:D9:4C secured 04
14:22:53.994 ->  7                    Coyote  6 -83dBm 78:8A:20:27:54:75 secured 04
14:22:53.994 ->  8                 CoyoteIoT  6 -83dBm 86:8A:20:27:54:75 secured 04
14:22:53.994 ->  9                CoyoteGast  6 -83dBm 82:8A:20:27:54:75 secured 04
14:22:53.994 -> 10             Vodafone-87A0 11 -93dBm 2C:3A:FD:6B:84:01 secured 04
14:22:53.994 -> current BSSID: 22:E8:29:A7:77:0D, strongest BSSID: 22:E8:29:A7:77:0D
14:22:53.994 -> Connecting from bssid:22:E8:29:A7:77:0D to bssid:22:E8:29:A7:77:0D, channel:1
14:22:53.994 -> WIFI_CONNECT_ONGOING
14:22:53.994 -> status=145 topic=PROTECTION-NO payload=0
14:22:56.957 -> status=140 topic=TOTAL-COMP-RUN payload=300
14:22:58.002 -> status=147 topic=OU-EEV1 payload=250
14:23:01.139 ->  connected to CoyoteIoT, IP address: 192.168.59.110 (-81dBm)
14:23:01.139 -> WIFI_CONNECT_OK
14:23:01.290 -> MQTTreconnect(): Attempting MQTT connection (MQTTclient.state=-1), WiFi.status()=3 ...
14:23:01.290 ->  connected
14:23:01.290 -> MQTTclient.connected=1
14:23:01.290 -> status=64 topic=connected payload=1
14:23:01.290 -> status=64 topic=Version payload=2.8
14:23:01.290 -> status=64 topic=RSSI payload=-81
14:23:01.290 -> status=64 topic=WIFI_LOST payload=0
14:23:01.290 -> status=64 topic=MQTT_LOST payload=0
14:23:01.290 -> status=64 topic=WIFI_BSSID payload=22:E8:29:A7:77:0D
14:23:01.290 -> MQTTreconnect(): 10 access points available
14:23:01.290 -> status=64 topic=fSCK payload=4231
14:23:01.290 -> status=64 topic=fMOSI payload=552
14:23:01.290 -> status=64 topic=fMISO payload=0
14:23:01.392 -> status=65 topic=Mode payload=Cool
14:23:01.392 -> power_status: off; received status_power: 0
14:23:01.392 -> status=64 topic=Power payload=Off
14:23:01.392 -> status=66 topic=Fan payload=Auto
14:23:01.392 -> status=67 topic=Vanes payload=?
14:23:01.392 -> status=71 topic=Tsetpoint payload=22.0
14:23:01.392 -> status=72 topic=Errorcode payload=0
14:23:01.450 -> status=135 topic=THI-R1 payload=15
14:23:02.292 -> status=136 topic=THI-R2 payload=39
14:23:03.194 -> status=137 topic=THI-R3 payload=14
14:23:04.133 -> status=134 topic=IU-FANSPEED payload=0
14:23:05.067 -> status=139 topic=TOTAL-IU-RUN payload=200
14:23:06.027 -> status=132 topic=OUTDOOR payload=6.25
[... 28 lines cut ...]
14:23:41.770 -> power_status: on; received status_power: 0
14:23:41.770 -> status=64 topic=Power payload=Off
14:23:41.863 -> status=148 topic=unknown payload=4306
14:23:41.863 -> Unknown operating data, MOSI_frame[DB9]=210 MOSI_frame[D10]=16
14:23:44.500 -> status=133 topic=THO-R1 payload=-4
14:23:52.023 -> status=148 topic=unknown payload=4306
14:23:52.023 -> Unknown operating data, MOSI_frame[DB9]=210 MOSI_frame[D10]=16
14:23:54.067 -> status=147 topic=OU-EEV1 payload=470
14:23:57.544 -> status=135 topic=THI-R1 payload=15
14:23:58.271 -> status=138 topic=OU-FANSPEED payload=4
14:23:59.394 -> status=137 topic=THI-R3 payload=14
14:24:02.292 -> status=144 topic=TDSH payload=0
14:24:05.893 -> status=142 topic=CT payload=0.27
14:24:07.291 -> status=148 topic=unknown payload=4306
14:24:07.291 -> Unknown operating data, MOSI_frame[DB9]=210 MOSI_frame[D10]=16
14:24:08.312 -> status=144 topic=TDSH payload=1
14:24:09.099 -> mhi_ac_ctrl_core.loop error: -1
14:24:21.622 -> status=133 topic=THO-R1 payload=-4
14:24:45.028 -> status=144 topic=TDSH payload=1
14:24:53.154 -> status=138 topic=OU-FANSPEED payload=0
14:25:12.510 -> status=136 topic=THI-R2 payload=38
14:25:13.517 -> status=144 topic=TDSH payload=0
14:25:17.121 -> status=133 topic=THO-R1 payload=-4
14:25:41.035 -> status=144 topic=TDSH payload=0
```

glsf91, 2024-01-14:
That is looking very good. Almost no mhi_ac_ctrl_core.loop errors.
Sometimes reflashing will help indeed. Sometimes I also flash first a zeroed bin file to clear the complete flash memory when I have strange behaviour.

You even can try the extended framesize if you want to get 3dauto and L/R vanes.

coyote1, 2024-01-14:
Yes, I had tried that. I had flashed the ESP several times, but always had the same problem. I also always selected "flash all contents".

Another thing I saw was that I didn't comment out the extended frame size, I just set it to false. Maybe that was my problem!
I will test further tomorrow and report back.

Übersetzt mit DeepL (https://www.deepl.com/app/?utm_source=ios&utm_medium=app&utm_campaign=share-translation)

glsf91, 2024-01-27:
@coyote1 Any update or can we close this one ?

coyote1, 2024-01-27:
Yes, we can close it. All three Aircons updated to 2.8 👍

closed 2024-01-27 by glsf91