absalom-muc/MHI-AC-Ctrl #182: Troom too high in my setup

KDC666, 2024-02-11:
Name of my AC: SCM 50 ZS + 2x SRK 20 ZS + SRK 25 ZS
Used MHI-AC-Ctrl HW: 3419d03006ddce74497c55a1a4aa63c1099d1669
Used MHI-AC-Ctrl SW: e85d4c3dcc641a77eca487e5964eee02fe52e80e

**Describe the bug**
First of all, hi and thanks for the effort you put into this nice project. At the moment I use my setup fro heating.

I setup my AC configuration with your hardware and software and added it to Home Assistannt today. It is now finally working and I'm really happy with it. But it is not flawless and I don't know why. The readback room temperature seem to have an akward offset and/or factor. 

Yesterday during my first tests I already noticed that the status of he room temperure is much higher (about 2 degree) than the setpoint and this was after a few hours. Having part of my setup working in Home Assistent I checked the room temperature curve during the night and I noticed, that it increased and was about 23°C this morning. When I went into the living room (where the unit is installed) I checked the temperature by manually reading my thermometer and it was at 19.7°C (I could even feel that 23°C was wrong ;-) ). So it seems there is some kind of offset or even negative factor.
I'm not sure if I missed anything but I just went through your documentation and code a second time an wasn't able to finy anything suspicious. 

[image]

As you can see in the picture I had to slightly adjust the strings to work with HA, apart from that, I didn't alter the code. 

[image]

The blue line is the room temperature, the purple one is the set value. At night the unit was turned off. Might this be the problem? Apart from that, even when it's on the measured temperature is too high.

If you have no idea why this is, I'll changed the calculation factor in the code to math the real temperature better, but this is more or less my last resort.

glsf91, 2024-02-11:
You are using the internal temperature sensor of the AC?
What is the value of OpData/Tsetpoint when you are use Tsetpoint of 19 degrees like in the picture?

KDC666, 2024-02-11:
Sorry, I forgot to mention this. Yes, it's the internal.

Like I said, not sure if this is a bug or if I'm doing something wrong. Unfortunatley, I just increased the temperature to the Troom is a lot closer to the set value now.

I just checked the OpData one and it is highter than /Tsetpoint as well:
[image]

glsf91, 2024-02-11:
This is not a bug indeed.
There can be a difference (a few degrees) between the temperature measured by the internal sensor and your own sensor in the room. That is because the heat will go to the ceiling of your room. So when measuring for example on the floor or at the ceiling will show a (big) difference.

The AC is using default an offset of 2 degrees. So if your setpoint is 21 degrees, the AC will use a setpoint of 23 degrees. OpData/Tsetpoint is the real setpoint used by the AC.
The AC is generating heat until the temperature of the internal sensor is the same as the OpData setpoint.
The offset is use to compensate between the temperature measured by the AC and the real room temperature.

Summary:
- Troom is the temperature of the internal sensor
- temperature in the room will be lower then Troom
- real setpoint used by AC is chosen setpoint + 2 degrees offset.

When using a central heating it will almost reach the exact temperature of the chosen setpoint on the thermostat. With an AC this is not mostly not the case. It can be possible for example to reach a room temperature of 19 degrees, you have to use a setpoint of 20 or 21 degrees. Do not compare this with central heating.

KDC666, 2024-02-11:
I'm fully aware of this, I have been using the AC for about 2 years now but controlled by hand (AC schedule of course too) and I know this.

But there is the temperature measured at night. The AC was completly off and the temperature stayed at the same, high value all night. And I'm pretty sure that it was much warmer in the room last night than this morning.

Nevertheless, I'll give it another try and see what happens today or tomorrow and get back with a result.

KDC666, 2024-02-12:
Okay, HA was running since yesterday afternoon and it captured some data:
[image]

I get the point with the general offset and I'm fine with that but what I still don't get:

Around 6:15pm the set temperature is turnt to 19°C  coming from 21°C and the actual temperature slowy drops as expected. Around 9pm the heating is turned off and from there on the room temperature increases and stays even higher as before (heating to 19°C). The room was definately colder than before as the insulation of the room isn't that good and we hat around 6°C outside. Has it something to do with the AC being turned off? Any ideas?

KDC666, 2024-02-13:
I think your're correct and this is not really a bug in your software but is related to the AC in some way. Unfortunately, I have no chance to add another 'smart' sensor.

The three Set values came from playing around with MQTT, I have two other units I added later when everything was working and they don't have them at all.
[image]

I'm pretty new to HA so no idea how to add the return air temp to the graph.

Here is another diagramm, this one is from my sons bedroom.
[image]

I tend to believe that it has something to do with hot air inside the AC not being blown away anymore and just cooling down over time. Or something like that.

The third room shows almost exactly the same behaviour.

glsf91, 2024-02-13:
I asked about the set to make sure there was no Troom under set (which is not the case as expected). 

Without an additional sensor it is hard to say. You can also try to put another analog temperature device on top of the AC. And before you switch on the AC in the morning take a look at the temperature and compare with the temperature of the internal sensor.
I also have never heard about this "problem" before. But I also find the temperature in de night "high".

henrykuijpers, 2024-02-16:
Would it be possible to add support for a ds18b20 for example which can be put in a more external location? 

That way, you can have influence over the temperature that is being used.

glsf91, 2024-02-16:
Support for a ds18b20 is already included. See bottom of hardware.md page. You have to enable this in the software.

glsf91, 2024-02-26:
@KDC666 Can we close this one ?

KDC666, 2024-02-26:
Yes, I think so, seems not to be an issue with the code/programm.

closed 2024-02-26 by glsf91