absalom-muc/MHI-AC-Ctrl #184: [W][component:214]: & [W][component:215]:

EvoSems, 2024-02-22:
:fire: Before you create an issue, please recheck the [Troubleshooting guide](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/Troubleshooting.md#troubleshooting)!

If your problem isn't mentioned in the [Troubleshooting guide](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/Troubleshooting.md#troubleshooting) please consider the following topics for your issue:
**To Reproduce**
Name of my AC:
Used MHI-AC-Ctrl HW: 
Used MHI-AC-Ctrl SW:

**Describe the bug**
I have a new AC. and am getting a lot of errors in the logs. is this something to worry about?
my other new ac. doesn't show these a=errors at all. no errors at all

`INFO Upload took 4.51 seconds, waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from lounge_ac.local using esphome API
WARNING Can't connect to ESPHome API for lounge_ac.local: Error resolving IP address: [Errno 8] nodename nor servname provided, or not known (APIConnectionError)
INFO Trying to connect to lounge_ac.local in the background
INFO Successfully connected to lounge_ac @ 192.168.31.81 in 0.036s
INFO Successful handshake with lounge_ac @ 192.168.31.81 in 0.377s
`
`[17:13:19][W][component:214]: Component <unknown> took a long time for an operation (0.06 s).
[17:13:19][W][component:215]: Components should block for at most 20-30ms.
[17:13:19][W][component:214]: Component <unknown> took a long time for an operation (0.06 s).
[17:13:19][W][component:215]: Components should block for at most 20-30ms.
[17:13:19][W][component:214]: Component <unknown> took a long time for an operation (0.06 s).
[17:13:19][W][component:215]: Components should block for at most 20-30ms.
[17:13:19][W][component:214]: Component <unknown> took a long time for an operation (0.06 s).
[17:13:19][W][component:215]: Components should block for at most 20-30ms.
[17:13:19][W][component:214]: Component <unknown> took a long time for an operation (0.06 s).
[17:13:19][W][component:215]: Components should block for at most 20-30ms.
[17:13:19][W][component:214]: Component <unknown> took a long time for an operation (0.06 s).
[17:13:19][W][component:215]: Components should block for at most 20-30ms.
[17:13:19][W][component:214]: Component <unknown> took a long time for an operation (0.06 s).
[17:13:19][W][component:215]: Components should block for at most 20-30ms.
[17:13:20][W][component:214]: Component <unknown> took a long time for an operation (0.06 s).
[17:13:20][W][component:215]: Components should block for at most 20-30ms.
[17:13:20][W][component:214]: Component <unknown> took a long time for an operation (0.08 s).
[17:13:20][W][component:215]: Components should block for at most 20-30ms.
[17:13:20][W][component:214]: Component <unknown> took a long time for an operation (0.06 s).
[17:13:20][W][component:215]: Components should block for at most 20-30ms.
[17:13:20][W][component:214]: Component <unknown> took a long time for an operation (0.06 s).
[17:13:20][W][component:215]: Components should block for at most 20-30ms.
[17:13:20][W][component:214]: Component <unknown> took a long time for an operation (0.06 s).
[17:13:20][W][component:215]: Components should block for at most 20-30ms.
[17:13:20][W][component:214]: Component <unknown> took a long time for an operation (0.06 s).
[17:13:20][W][component:215]: Components should block for at most 20-30ms.
[17:13:20][W][component:214]: Component <unknown> took a long time for an operation (0.06 s).
[17:13:20][W][component:215]: Components should block for at most 20-30ms.
[17:13:21][W][component:214]: Component <unknown> took a long time for an operation (0.06 s).
[17:13:21][W][component:215]: Components should block for at most 20-30ms.
[17:13:21][W][component:214]: Component <unknown> took a long time for an operation (0.06 s).
[17:13:21][W][component:215]: Components should block for at most 20-30ms.
[17:13:21][W][component:214]: Component <unknown> took a long time for an operation (0.06 s).
[17:13:21][W][component:215]: Components should block for at most 20-30ms.
`

p.s thanks for all your hard work

glsf91, 2024-02-22:
You are using eps home. I think from https://github.com/ginkage/MHI-AC-Ctrl-ESPHome. 
You should post the issue there. So please close this one.

EvoSems, 2024-02-22:
apologies. you are right. I had both tabs open doh

closed 2024-02-22 by EvoSems