absalom-muc/MHI-AC-Ctrl #186: Can't send data to the ac "SRK20ZS-W"

leMax6608, 2024-05-04:
**To Reproduce**
Name of my AC: SRK20ZS-W
Used MHI-AC-Ctrl HW: Wemos D1 Mini from AZDelivery
Used MHI-AC-Ctrl SW: 2.8

**Describe the bug**
A clear and concise description of what the bug is.
Hello. Today I tried to make thze climate control with a wemos d1 mini. The data from the ac is showed in IOBroker but I just can't figure out how to send data. Everything I tried failed. 
[image]
[image]
[image]
[image]
There are pictures of my MQTT cofiguration and the installation in the ac.

Can you please help me fixing this problem?

absalom-muc, 2024-05-04:
What is your configuration for
`MQTT_SET_PREFIX MQTT_PREFIX` ?
Per default it is "set/", not "Set/"

leMax6608, 2024-05-05:
I think you‘re right. I will try it this evening and let you know if it works.

leMax6608, 2024-05-08:
Hi. I tried it and just the /set/Power won‘t work. But when I enabled Power On when changing Modes, it suddenly works. I think I will do a YT Video on how to do this whole process and set it up with home assistant.

absalom-muc, 2024-05-11:
Thanks for the feedback

closed 2024-05-11 by absalom-muc