absalom-muc/MHI-AC-Ctrl #187: RFI: Reason for checking TD for value 0x12

bassduh, 2024-05-16:
**To Reproduce**
Name of my AC: SRK25ZS-W
Used MHI-AC-Ctrl HW:Wemos D1 mini
Used MHI-AC-Ctrl SW:2.8

**Describe the bug**
Not sure if it is a bug, I guess it is a deliberate consideration, but I do not understand it yet :-)

What is the reason for checking if the value is less than 18 (0x12) and return a string "<=30"?

```
case opdata_td:
  if (value < 0x12)
	strcpy(strtmp, "<=30");
  else
	itoa(value / 2 + 32, strtmp, 10);
  output_P(status, PSTR(TOPIC_TD), strtmp);
```

https://github.com/absalom-muc/MHI-AC-Ctrl/blob/6092479541e22568e9743790d6bf7b203b4cbd6a/src/MHI-AC-Ctrl.ino#L397

Maybe I oversee something, but this value could output a value less than 41 to the MQTT topic, right?
Now I see some issues in Home Assistant, because it can't parse "<=30" to a numeric, so the value at its lowest is 41.

absalom-muc, 2024-05-17:
For the interpretation of the opcodes I compared the AC internal opcode values with the output of a commercial available adapter. And for <0x12 the output was "<=30".

glsf91, 2024-05-19:
Lowest value is indeed 18/2 + 32 =41 (otherwise "<=30"). 
Please close the topic if the answers are satisfied.

bassduh, 2024-05-19:
Thanks for your responses, I'll give it a shot in either the Hame Assistant side (mqtt template) or a small change in the ino-file.

For now I'll close the issue.

closed 2024-05-19 by bassduh