absalom-muc/MHI-AC-Ctrl #188: MHI SRK20MA-S model

MrC0nst, 2024-05-19:
Good day, everyone!
I am trying to connect my AC to mqtt. But faced with some problems. 

**To Reproduce**
Name of my AC: MHI SRK20MA-S _(indoor unit)_ / SRC20MA-S _(outdoor unit)_
Used MHI-AC-Ctrl HW: Wemos D1 mini (8266), ESP32 Devkit. With and without level shifter (4 channels, bi-directional. Same as in BOM file)
Used MHI-AC-Ctrl SW: MHI-AC-Ctrl v2.8

**Describe the bug**
The AC board initially did not have CNS connector but this was easily resolved with iron:
[image]
[image]

All componets from bottom side of PCB are on place. Motherboard almost same as https://github.com/absalom-muc/MHI-AC-Ctrl/issues/149 

**First problem is low SCK frequency 2720Hz**
```
Starting MHI-AC-Ctrl v2.8
CPU frequency[Hz]=80000000
ESP.getCoreVersion()=3.1.2
ESP.getSdkVersion()=2.2.2-dev(38a443e)
ESP.checkFlashCRC()=1
Measure frequency for SCK, MOSI and MISO pin
SCK frequency=2720Hz (expected: >3000Hz) out of range!
MOSI frequency=392Hz (expected: <SCK frequency) o.k.
MISO frequency=0Hz (expected: ~0Hz) o.k.
```
**The second problem is that I cannot control the air conditioner, but only receive data from it** 
[image]

Some debug logs:
[console_framesize20.txt](https://github.com/absalom-muc/MHI-AC-Ctrl/files/15370068/console_framesize20.txt)
[console_framesize33.txt](https://github.com/absalom-muc/MHI-AC-Ctrl/files/15370069/console_framesize33.txt)
[mhi-ac-trace.txt](https://github.com/absalom-muc/MHI-AC-Ctrl/files/15370070/mhi-ac-trace.txt)
[spi_logger.txt](https://github.com/absalom-muc/MHI-AC-Ctrl/files/15370071/spi_logger.txt)

Additional devices available:
- 24mhz 8 channel logic analyzer (saleae clone)
- Multimeter
- Oscilloscope

Technical manual for my AC:
[Technical_Manual_srk20mas.pdf](https://github.com/absalom-muc/MHI-AC-Ctrl/files/15370072/Technical_Manual_srk20mas.pdf)

glsf91, 2024-05-19:
Only try first framesize 20. You have already a lot of errors there (but with 33 a lot more). 
Take a look with the logic analyser if the frames/bytes looks right. Here you can find a description of the timings: https://github.com/absalom-muc/MHI-AC-Trace/blob/main/SPI.md

If you use saleae Logic software, add an SPI analyzer and you can use the following settings:
[image]

The settings for MOSI, MISO and Clock are depending on your channel connection with the board of course.
In the picture you see the first 3 bytes of the frame. 

MOSI is coming from AC. MISO is created by the mhi-ac-ctrl board/software.
Try to capture from the start of the software (hold reset, start capture and release reset after a few seconds) . Also capture debug logging at the same time. 
Hopefully it will pickup the first frame at the right point. It takes sometime some retries :-)

MrC0nst, 2024-05-19:
I acquired this AC in spring 2014, but this is year 2011 model (based on motherboard labels and technical documentation from 1st post). Maybe protocol in this AC is a test | draft version and not all attributes and values are encoded-decoded right?
[image]

I noticed that when setting the fan speed using the IR remote control, the data is almost instantly displayed in the MQTT. But if you change other settings, there may be a delay of up to 6 seconds to update values.

[log1.log](https://github.com/absalom-muc/MHI-AC-Ctrl/files/15371040/log1.log)
[log1.sal.txt](https://github.com/absalom-muc/MHI-AC-Ctrl/files/15371041/log1.sal.txt)
[log2.log](https://github.com/absalom-muc/MHI-AC-Ctrl/files/15371042/log2.log)
[log2.sal.txt](https://github.com/absalom-muc/MHI-AC-Ctrl/files/15371043/log2.sal.txt)

glsf91, 2024-05-20:
Please compile for a CPU frequency of 160MHz instead of 80 MHz and try again. Do not use a ESP32 if this is the case.
Check if the SCK frequency is not out of range anymore or there are less -4 errors.

Do you also use the PCB from Hardware.md ? If no, please show a picture of your "PCB".

I see the timings are a bit different then my AC (SRK xx ZS-W). 
A time ago some stuff with sending MISO has changed. So you can also give V2.6 with the old MISO sending stuff a try if this will do a better job for you.

glsf91, 2024-05-21:
That's a pitty.
It is better to use a level shifter on MISO, MOSI and SCK lines. 
Adding this print debug can cause more errors because of the timing by the way.

The ESP is following the SCK line to determine the MOSI byte (and create the MISO byte). 
The -4 error indicates it is not detecting an expected falling edge on the SCK line on time in the frame. That means the SCK is detected "damaged". Maybe there is some noise on the SCK line.
Maybe you can see this with an Oscilloscope.

In the logic analyzer file from previous post I saw the SCK is a lot if times only 5 uS low. With my AC this is 15-20uS. So maybe this is too critical.
Detecting the SCK frequency at boot is done with interrupts (different from normal operation). When I look into your logic analyzer file you should see a SCK freq of about 2880 Hz. So you are missing rising edges because of noise or timing to critical.

glsf91, 2024-05-25:
> Hi glsf91
>
> *It is better to use a level shifter on MISO, MOSI and SCK lines.*
> I redesigned the board and added back the level shifter:
> board.jpg (view on web)
> also mute function was added (I'll check later). Moved back to v2.8
>
> Currently data exchange only works in one direction from AC to ESP.
> If I try to change the air conditioner settings through the Home
> Assistant, nothing happens. Only an error appears in the MQTT
> mqtt.png (view on web)
> How to determine whether the request reaches the air conditioner or not?
> Which module responds with error 221 cmd_received = invalid parameter?

Error means your mqtt command is not right. In this case fan_only is not
allowed. See sw_configutation.md

MrC0nst, 2024-05-25:
Oscilograms:

### SCK
[image]
[image]
[image]
[image]

### MOSI
[image]
[image]
[image]

### MISO
[image]
[image]
[image]
[image]

### SCK (1-Y) and MOSI(2-G)
[image]
[image]

### SCK(1-Y) and MISO(2-G)
[image]
[image]
[image]
[image]

Seems like MISO from AC side is pulled to 3v because unconnected pin HV2 has 5V from ESP8266 side. I need to find out whether the air conditioner receives signal packets from the ESP. Is there any way to create a deliberately incorrect packet so that the air conditioner reports an error in any way?

glsf91, 2024-05-25:
> Oscilograms:
> SCK
>
> sck_1.jpg (view on web)
> sck_2.jpg (view on web)
> sck_3.jpg (view on web)
> sck_4.jpg (view on web)
> MOSI
>
> mosi_1.jpg (view on web)
> mosi_2.jpg (view on web)
> mosi_3.jpg (view on web)
> MISO
>
> miso_1.jpg (view on web)
> miso_2.jpg (view on web)
> miso_3.jpg (view on web)
> miso_4.jpg (view on web)
> SCK (1-Y) and MOSI(2-G)
>
> sck_mosi_1.jpg (view on web)
> sck_mosi_2.jpg (view on web)
> SCK(1-Y) and MISO(2-G)
>
> sck_miso_1.jpg (view on web)
> sck_miso_2.jpg (view on web)
> sck_miso_3.jpg (view on web)
> sck_miso_4.jpg (view on web)
>
> Seems like MISO from AC side is pulled to 3v because unconnected pin HV2
> has 5V from ESP8266 side. I need to find out whether the air conditioner
> receives signal packets from the ESP. Is there any way to create a
> deliberately incorrect packet so that the air conditioner reports an error
> in any way?

As far as I know the ac will not report an error when sending a wrong
packet.
What do you mean with an unconnected hv2 pin?
According to the hardware design there is no unconnected pin.

MrC0nst, 2024-05-25:
_What do you mean with an unconnected hv2 pin?_
If you cut wire X1-2_J2.2 at the red mark, then at pin J2.2 there will be a signal with an amplitude of 5 volts. Or connect J1.6 to JP1.5 with a blue marked wire and take measurements at pin J2.6 (Connected it to make sure the level shifter is working correctly)
[image]

With default circuit i have MISO amplitude of 3 volts only. Measured at MSIO PIN on X1 Connector (X1-2)
[image]

MrC0nst, 2024-05-26:
After switching to the level shifter using an optocoupler, the voltage amplitude now reaches 5 volts. But the air conditioner does not follow commands. What additional steps can be taken to establish two-way communication?

SCK(1-Y) and MISO(2-G)
[image]
[image]

glsf91, 2024-05-26:
what happens if the ac is turned on bij de rc and you send the power off
command to mqtt?
Also check the topic cmd_received says ok after command to make sure your
command is ok. Or check debug log.

MrC0nst, 2024-05-26:
_what happens if the ac is turned on bij de rc and you send the power off
command to mqtt?_
Nothing changes. The air conditioner continues to operate, the command status is `cmd_received = o.k.`
[image]
[image]

glsf91, 2024-05-26:
hmmm. maybe because your ac is that old and had no connector on the pcb, it
has no option to receive commands.
Or has a different MISO.
Normally you take another MHI device and take a look with a logic analyzer.
But thats is not possible here.

MrC0nst, 2024-05-28:
I think the issue can be closed. Data can only be read from the air conditioner. I will control my AC by sending a signal directly to the TSOP data pin.
PS: IR protocol is not reverse engineered yet. Well known are RKX502A001C remote (88 bit) and RLA502A700B remote (152 bit) https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/ir_MitsubishiHeavy.h Mine is 106 bit