absalom-muc/MHI-AC-Ctrl #189: FDC100VNP support

matheuscanela, 2024-05-22:
This is not an issue, but I couldn't find an option to raise a discussion.

Name of my AC: FDC100VNP

**Describe the bug**

I don't have access yet to my unit, but I am not sure if this project supports it, is there any way to know if it is supported? 
I contacted Mitsubishi and they recommended the device below

[MH-RC-WIFI-1B_Install Manual.pdf](https://github.com/SwiCago/HeatPump/files/15399135/MH-RC-WIFI-1B_Install.Manual.pdf)

glsf91, 2024-05-22:
FDC100VNP is the outdoor unit. 
The module needs to be connected to the indoor unit. So it depends on the type of the indoor unit I think.

matheuscanela, 2024-05-31:
Thanks folks!

I don't have access to the roof yet and it may take a few months :( 

I had a chat with the company, and they said that the indoor unit may be one of these

[image]

vergoossen, 2024-06-10:
Also interested in wifi for the fdtc series.
Have already some wifi modules on the skr's.

The original wifi module for the fdtc series is the IntesisBox INWFIMHI001R100. For the skr's the original wifi module is also Intesis i think.

Anyone which can help with this?

matheuscanela, 2024-06-11:
Hey @vergoossen

Yes, the manual provided by Mitsubishi talks about Intesis, have you used it? Does it integrate well with HA?

I am waiting for a few changes in my ceiling to be able to check the machine details as it is not easy to access.

glsf91, 2024-06-11:
> The original wifi module for the fdtc series is the IntesisBox INWFIMHI001R100. For the skr's the original wifi module is also Intesis i think.

This wifi model is not for SRK. It is also connected with 2 wires. So it cannot be replaced by mhi-ac-ctrl.

vergoossen, 2024-06-11:
This particular Intesis model not indeed. I meant there’s another Intesis model for the SRK.

glsf91, 2024-07-18:
closed because of no activity anymore

closed 2024-07-18 by glsf91