absalom-muc/MHI-AC-Ctrl #190: Power reading stays at zero

persistive, 2024-05-28:
First my gratitude for this fantastic piece of hard and software.
Everything seems to work fine exept the power reading in kWh stays at zero.
The current reading is ok.
Is this a known issue?

Greetings,

Paul

glsf91, 2024-05-29:
This is not a known issue. Maybe not all AC's is supporting this.
I have a SRK35ZS-W which is working.
It is only working when the AC is operating.

persistive, 2024-05-29:
Mine is type SRK35ZS-WT (indoor) and type SCM60ZS-W (outdoor unit).
The same as you have.
Everything works except the kwh reading.
To be honest, i make use of the esphome version, but as far as i know there is no difference in functionality.
Thanks for your feedback

glsf91, 2024-05-29:
There can be a difference when using the esphome software. So try if
possible the original sw from here to check if it is still 0 kWh after
operating for some time.
If you cannot try the original sw I would suggest to create an issue at the
place of the esphome variant. Or take a look at a similar issue
https://github.com/ginkage/MHI-AC-Ctrl-ESPHome/issues/81

You also have a multisplit I think. I have a single unit. I don't know if
that is making a difference.
I have not seen your issue before.

glsf91, 2024-06-09:
Closed because of no activity anymore.

closed 2024-06-09 by glsf91