absalom-muc/MHI-AC-Ctrl #194: Support for SRK13CHV?

ferbulous, 2024-07-16:
Hi, not a bug
I have a question regarding the CNS port 
For my MHI (SRK13CHV-4), that port isn’t listed in the diagram and I couldn’t remove the pcb without some downtime

[image]
[image]

I googled the model number instead and apparently there’s CNS labeled with no JST connector

[image]

Am i correct to assume that port is still functioning and all I need to do is just solder JST connector to it?

glsf91, 2024-07-16:
Only somebody with the same unit and did the same can tell you.
There will be a reason the connector is not there. But you can give it a try.

ferbulous, 2024-07-16:
Just noticed this post here where the pins just need to be soldered

https://github.com/absalom-muc/MHI-AC-Ctrl/issues/143#issuecomment-1510408359

Unfortunately it looks like my unit has 6 pins instead of 5

Dennisjvdb, 2024-08-08:
Would it be possible to design a PCB with a 6 pin connector? So the models with CNE port or 6pin CNS port can also be supported?

ferbulous, 2024-08-10:
I'd say the better question is if we could send a replacement pcb with a 6 cns pin to the dev @absalom-muc  if he could work with that.

Dennisjvdb, 2024-08-10:
I think al models have a 6-pin cne connector. Might be better as no soldering would be required.

absalom-muc, 2024-08-11:
When my understanding is correct we don't know the technical differences between the 5 and 6-pin connector, e.g.

1. Can the 6-pin connector be used for a remote control? (someone could check if there exists a commercial available remote control for this.)
2. What is the protocol used for the 6-pin connector? (could be also MHI-SPI-Ctrl's SPI, or could be completely different)
3. If it is SPI what is the purpose of the 6th pin and what is the assignment of the other pins

btw: I don't plan adaption for a 6-pin version

Dennisjvdb, 2024-08-11:
In [this](https://github.com/dvisser/MHI-ESP-PCB) project, the CNE connection is used for added features. So, I think it can be used for remote control. I'm not sure what the protocol is, as I have a limited understanding of these protocols. If the same protocol is used and we know the pin assignment, would it theoretically be possible to connect this module to a 6-pin male connector without using the 6th pin?

absalom-muc, 2024-08-11:
[Here](https://github.com/dvisser/MHI-ESP-PCB/blob/main/Images/Module%20-%20Top.jpg), I see only a 5-pin connector. Can't see if it is the usual 5-pin CNS or something different.
And I can't find an accessable schematic (I don't use Kikad)

Dennisjvdb, 2024-08-11:
I think [this](https://github.com/dvisser/MHI-ESP-PCB/blob/main/MHI%20ESP.pdf) the schematic in pdf.

absalom-muc, 2024-08-11:
Ah, o.k., but no support of 6-pin:
[image]

glsf91, 2024-08-19:
I think we can close this one. Otherwise I can reopen it if needed.

closed 2024-08-19 by glsf91