absalom-muc/MHI-AC-Ctrl #197: Energy efficiency monitoring

MalteSchm, 2024-08-24:
Regulations in Germany require energy efficiency monitoring to get heat pump government founding. The German distributor of MHI devices has a solution that fulfills this requirement. This solution connects to the SPI port (*) of all IU to accomplish this. Looking at the solution this seems to work on the data that is also used by MHI-AC-Ctrl. I'm trying to understand if this can be realized with MHI-AC-Ctrl. Initially I thought that the 

1. Room temperature
2. Temperatur of the returned (heated or cooled) air
3. Volumetric air flow, calculated via fan speed lookup

could be used to calculate the energy transferred by a single IU. This could be used together with an energy meter to measure efficiency. Having build and installed an MHI-AC-Ctrl I believe that 2) (Temp of returned air) is not available. Is this understanding correct? Are there maybe also other ideas to accomplish this?

(*) My understanding from reading the manual is that they convert the data to the X/Y bus to bring it into a device that sends this to the cloud. Energy efficiency monitoring is then provided by this cloud service.

glsf91, 2024-08-24:
I think returned air is available as OpData/RETURN-AIR

MalteSchm, 2024-08-24:
Is room temp then also returned?
I saw Troom and RETURNED-AIR but both values were almost the same

glsf91, 2024-08-24:
yes, that is Troom.

MalteSchm, 2024-08-25:
Troom and RETURN-AIR are both driven by the same temperature sensor that sits here:
[image]

I did a quick fan / heat test and both values changed

glsf91, 2024-08-25:
We can not do better then this.
This is available through opdata: [doc](https://www.hrponline.co.uk/media/pdf/41/42/ed/Beijer-Ref-Service-Support-Handbook-19cWKESQUhzVIy5.pdf#page=9)

MalteSchm, 2024-08-27:
> We can not do better then this. This is available through opdata: [doc](https://www.hrponline.co.uk/media/pdf/41/42/ed/Beijer-Ref-Service-Support-Handbook-19cWKESQUhzVIy5.pdf#page=9)

Understood. I also like to say that I'm not asking for a new feature. I can implement this myself if I'd only knew the approach

> Do you have a link to the device and manual ?

You mean the indoor unit? That is a SRK-15ZS-WF

glsf91, 2024-08-27:
There is no more available then the opdata. So you or we cannot add more.

No, a link to the device and manual of the german distributor.

MalteSchm, 2024-08-27:
I'm using 
Outdoor unit: [SCM41ZS-W](https://www.s-klima.de/shop/sx-serie/sx-aussengeraete/aussengeraet-scm41zs-w.html) 
Indoor units: [SRK-15ZS-WF](https://www.s-klima.de/shop/sx-serie/sx-wandgeraete/wandgeraet-srk15zs-wf-1-50-kw.html)

The required components that enable energy efficiency monitoring are:
[BIKN2-E](https://www.s-klima.de/shop/zubehoer/fernbedienungen-mhi/adapterplatinen/adapterplatine-sc-bikn2-e.html) - This converts the SPI bus interface into an bus called "X/Y bus"
[Comptrol-Web](https://www.s-klima.de/shop/comptrol/comptrol-erweiterungsmodule/comptrol-4web-4webnrc.html) - Connects to the X/Y bus and the internet to provide data to the cloud
[e-Comptrol](https://dam.stulz.com/pinaccess/pinaccess.do?pinCode=GqzCmxSgIzru) - Is the webinterface

I liked the diagram on page 6-12 in the doc shown on this page:
https://www.fachklima24.de/mitsubishi-heavy-ct4web-e-comptrol-4web-steuerung-und-ueberwachung.html?language=de

Overview: https://www.s-klima.de/beg-foerderung/
Details: https://www.s-klima.de/beg-foerderung/auswahl-waermepumpen/

This effort is insane in my mind. It is very likely driven by the subsidy. This setup allows selling an overpriced extension to re-direct some of the money.
It annoys me that the only data source is the SPI bus. The Wifi module should have all the data and is internet connected and I feel that I'd be ripped off by purchasing these items.

absalom-muc, 2024-08-28:
@MalteSchm : Are you aware that there is some additional HW docu related to Operation Data [here](https://github.com/absalom-muc/MHI-AC-Trace/blob/71d33871540fb190d76835627bdbc3bfcb0848b2/SPI.md)?
I can't imagine that for BAFA/KFW a self-made solution for energy monitoring is acceptable since it is not protected against manipulation.

MalteSchm, 2024-08-28:
Yes I am aware of this document.
I also checked some MHI pipeing diagrams. They are clear regarding the different temperature measuring points. They also use the same names.
While looking at older pipeing diagrams it became clear to me that MHI has changed the system configuration over time. This explains why some measurements do not provide useful info. 

Right now I believe the the EEV pulse rate is used to derive the coolant flow. This flow rate carries a certain latent energy. The sum of the energies of all indoor units divided by the used energy of the outdoor unit is the efficiency 

If I'd only knew how EEV pulse rate converts into coolant flow...

glsf91, 2024-08-30:
I cannot add more to this I think.
I expect all data which is used in that external device should also be available in mhi-ac-ctrl. You have to figure out what to use.

glsf91, 2024-09-27:
I think we can close this one. Otherwise let me know

closed 2024-09-27 by glsf91