absalom-muc/MHI-AC-Ctrl #199: SRR series not working

Dduck112, 2024-10-19:
Hi all, I orders the board on Nodo-Shop and included the soldering service.
I have a MHI srr35zs-w with the required CNS connection, 5 pins.
Installed the ESPHome code and read out the log wireless while on usb power.
Turned the power from the airco down, and connected the board. Turned it back on, but the ESP did not get powered.
I tried pressing the reset button and watched for the little blue light, but it did not light up.
(when I have the ESP connected with a usb power, it does light up.)
So I ques the SRR are not supported :(  (I did not dare to power it with a usb while connected too the airco)

glsf91, 2024-10-20:
Check if the TSR2450 SUPPLIES 5V to the wemos. If not check if the AC supplies 12V to the TSR

Dduck112, 2024-10-20:
Hi,
I get 12.3V on the TSR and 4.9 to the wemos.
So no idea why it does not boot up at all.

fonske, 2024-10-20:
Does your esp still work on USB source only? Or is it dead on usb?

I see that the pcb from node-shop is missing the larger capacitor as in the original design?

Dduck112, 2024-10-20:
Hi,  I disconnected the board from the airco, and powered it using a 12v power supply, the wemos did not turn on...
I tried a other wemos, and it did  turn on. I flashed the bin downloaded from ESPhome in HA to the new board and added the code from the examle to the main code. tried to connect with the wireless log, and it worked. Connected it back to the airco and it is powering the wemos :)  I can read the log from within HA... see below, But all entities are unavailable.

side note, it is my first ESPhome node. So not much experience here. 

Log:
INFO ESPHome 2024.10.0
INFO Reading configuration /config/esphome/espairco.yaml...
INFO Starting log output from 192.168.2.40 using esphome API
INFO Successfully connected to espairco @ 192.168.2.40 in 0.063s
INFO Successful handshake with espairco @ 192.168.2.40 in 0.396s
[14:09:35][I][app:100]: ESPHome version 2024.10.0 compiled on Oct 20 2024, 13:51:24
[14:09:35][W][component:237]: Component MhiAcCtrl took a long time for an operation (53 ms).
[14:09:35][W][component:238]: Components should block for at most 30 ms.
[14:09:36][W][component:237]: Component MhiAcCtrl took a long time for an operation (54 ms).
[14:09:36][W][component:238]: Components should block for at most 30 ms.
[14:09:36][W][component:237]: Component MhiAcCtrl took a long time for an operation (52 ms).
[14:09:36][W][component:238]: Components should block for at most 30 ms.
[14:09:36][W][component:237]: Component MhiAcCtrl took a long time for an operation (53 ms).
[14:09:36][W][component:238]: Components should block for at most 30 ms.
[14:09:36][W][component:237]: Component MhiAcCtrl took a long time for an operation (52 ms).
[14:09:36][W][component:238]: Components should block for at most 30 ms.
[14:09:36][W][component:237]: Component MhiAcCtrl took a long time for an operation (52 ms).
[14:09:36][W][component:238]: Components should block for at most 30 ms.
[14:09:37][W][component:237]: Component MhiAcCtrl took a long time for an operation (51 ms).
[14:09:37][W][component:238]: Components should block for at most 30 ms.
### This go's on and on

fonske, 2024-10-20:
you can use this in your code
```
logger:
  baud_rate: 0
  logs:
    esp32.preferences: DEBUG
    sensor: DEBUG
    text_sensor: DEBUG
    switch: DEBUG
    button: DEBUG
    number: DEBUG
    climate: DEBUG
    component: ERROR
```

glsf91, 2024-10-20:
Please continue in the repo with this issue where you got the mhi esphome sw.
They can support esphome issues better there. I will close this one.

closed 2024-10-20 by glsf91