absalom-muc/MHI-AC-Ctrl #202: Manufacturer Wifi units wiring question

ThomasHFWright, 2024-12-05:
Name of my AC: SRK35ZS-WF
Used MHI-AC-Ctrl HW: Nodo shop ESPAC Wifi interface MHI-AC-Ctrl +WeMos D1 Mini (https://www.nodo-shop.nl/en/our-products/228-espac-wifi-mitsubishi-heavy-industries-interface.html)
Used MHI-AC-Ctrl SW: N/A

**Describe the bug**
Has anyone tested the cable that connects to the manufacturer wifi units to check the orientation of the cables are the same as expected by the hardware design in this repo?
My installer didn't have access to non-wifi units so just gave me a discount to get the same price on the units with wifi units pre-installed.
They have installed the units without the proper spacing to be able to open them up and allow me to swap cables (getting this fixed with the aftercare service).
In the meantime I want to see if I can just splice the manufacturer cable to connect to the module I bought from Nodo shop but I don't want to risk frying my aircon units or esp hardware.

See picture of the wifi unit:
[image]
I've identified which pins are the 12V and GND but not sure about the 3 middle pins as I couldn't find any obvious trace to what I expect are the MISO, MOSI and SCK pins on the chip.
I'm no electronics expert so wanted to ask if there's any risk to my units if I just try those pins randomly until I can identify the correct orientation. Thoughts?

fonske, 2024-12-05:
This cable has the same wiring order as the MHI-AC-CTRL
I checked already.

I was planning to make a pcb with this connector on the board too
[image]
[image]

ThomasHFWright, 2024-12-05:
That's awesome, do you know what the connector standard is? I'd love to just buy or make a converter from the manufacturer cable to the one on the piece of hardware I bought

fonske, 2024-12-05:
@ThomasHFWright 
I think the connector is a SM05B-GHS-TB(LF)(SN) from jst

glsf91, 2024-12-05:
I cut the wifi cable in the middle (power off AC !!). Then bought this: https://www.aliexpress.com/item/1005003977653324.html
Cut 1 of the cables in the middle. Soldered 1 part to the wifi cable from AC. Other part to the wifi module. Used Heat Shrink Tube to protect.
Now I can connect (with AC mains power off!) the mhi-ac-ctrl module or wifi module to the AC.

fonske, 2024-12-05:
@glsf91 
Are you sure you this fits on this module, because the pitch size (1,25) looks smaller then the one in your link?

ThomasHFWright, 2024-12-05:
It looks like these might be the right size https://www.aliexpress.com/item/1005006135792829.html

glsf91, 2024-12-05:
I used the cable from my link. This cable (JST-XH) fits on the mhi-ac-ctrl module like described in https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/Hardware.md under connector. It also fits the CNS port on the AC.
It won't fit of course the connector on the pcb of the wifi module. 
But I see your cable of the wifi module is short. So probably makes not much sense to cut the cable like I did. My cable was 1 meter.

ThomasHFWright, 2024-12-05:
> @ThomasHFWright I **think** the connector is a SM05B-GHS-TB(LF)(SN) from jst it doesnt fit on the standard soldering connection though because of 1.25 mm pitch instead of 2 mm
> 
> I am happy to make one for you, when i order the new pcb's

I appreciate the thought, but I already have all the hardware for the regular JST-XH cable so just looking for a simple adapter to put together

fonske, 2024-12-05:
@ThomasHFWright 
Not sure if it will fit on the original connector as it is GH
https://www.lcsc.com/datasheet/lcsc_datasheet_2304140030_JST-SM05B-GHS-TB-LF-SN_C189896.pdf

ThomasHFWright, 2024-12-05:
> @ThomasHFWright Not sure if it will fit on the original connector as it is GH https://www.lcsc.com/datasheet/lcsc_datasheet_2304140030_JST-SM05B-GHS-TB-LF-SN_C189896.pdf

hmm yes you're right, I can't see any listings with female connectors. Only male such as https://www.aliexpress.com/item/1005007295303376.html
I suppose I can just splice the cable while it remains plugged in as long as I cut the power.

ThomasHFWright, 2024-12-05:
Anyone know where to look for a 5 pin JST GH breakout board? Closest I could find is a 6 pin https://mktechnic.com/product/jst-1-25-6-pin-female-connector-adapter-pcb-breakout-board/

ThomasHFWright, 2024-12-05:
I ended up ordering some little boards on JLCPCB to plug into and keep everything stock.
[image]
[image]

[PickAndPlace_PCB_JST-GH-5-pin-breakout_2024-12-05.csv](https://github.com/user-attachments/files/18026016/PickAndPlace_PCB_JST-GH-5-pin-breakout_2024-12-05.csv)
[Gerber_JST-GH-5-pin-breakout_PCB_JST-GH-5-pin-breakout_2024-12-05.zip](https://github.com/user-attachments/files/18026017/Gerber_JST-GH-5-pin-breakout_PCB_JST-GH-5-pin-breakout_2024-12-05.zip)

fonske, 2024-12-18:
@ThomasHFWright 
I got my pcb's also delivered today,
The connector is the correct one.

[image]

glsf91, 2025-01-12:
I close this one. We can reopen if needed

closed 2025-01-12 by glsf91