absalom-muc/MHI-AC-Ctrl #203: Mitsubishi WF-RAC Interface wiring

tech2mar, 2024-12-12:
Hi friends,

Is it possible that this Mitsubishi WF-RAC Interface module has wiring issues? It seems like it’s not powering on, as the device doesn’t work at all. Could it be that the 12V+ and GND connections are swapped or incorrectly wired? Has anyone experienced something like this before?

I recently made some PCBs for my home and sold this module to a friend, but he’s struggling to get it to work. Unfortunately, I haven’t had the chance to use this module myself, as I prefer a local solution for my own unit.

Any advice or insights would be greatly appreciated!

Very strange.

[image]

[image]

glsf91, 2024-12-13:
You can see [here](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/Hardware.md#connectorl) how the board is connected to he AC. So check if your cable is doing the same.
You can also check if the Wemos receives 5V from the TSR_1-2450.
If that is the case take a look at the serial logging of the Wemos with a usb cable (and the module connected to the AC).

tech2mar, 2024-12-19:
Update maybe the Modul is not working at all, is it possible to flash the Modul and using It without MQTT or Home Assistent? With Esphome and a direct Esphome server on it?

glsf91, 2024-12-20:
You can use the module without MQTT. See CONTINUE_WITHOUT_MQTT  in support.h
But if you don't use an external DS18x20 temperature sensor it has no sense to use the module without MQTT.
For esphome you have to use: https://github.com/ginkage/MHI-AC-Ctrl-ESPHome
(you can ask question about esphome there)

tech2mar, 2024-12-22:
Thank you for your help.

glsf91, 2025-01-12:
I close this one. We can reopen if needed

closed 2025-01-12 by glsf91