absalom-muc/MHI-AC-Ctrl #204: Usage without MQTT (no issue just question)

Haegar88, 2025-01-02:
Hi,
i don't have/need/know MQTT. I just just want to Start/Stop the device via HTML-Page of the ESP or depending the power of the solar inverter. Can someone please help me with some small framework? Can I call on Button click mhi_ac_ctrl_core.set_power(power_on/off)?
How can i register the Status-Handler?
Many thanks in advance.
BR Haegar

glsf91, 2025-01-03:
You can find a lot of examples of adding a webpage here: https://randomnerdtutorials.com/projects-esp8266/
I would advise to use this one: https://randomnerdtutorials.com/esp8266-nodemcu-async-web-server-espasyncwebserver-library/
You can indeed call mhi_ac_ctrl_core.set_power(power_on/off)

I don't understand what you mean exactly with "How can i register the Status-Handler"

Haegar88, 2025-01-03:
Hello, it's not about how I make the website and the surrounding, but I would need some tips on how to use the functions from the MHI-AC-Ctrl. I want to make my own lean sketch without all the MQTT stuff and call the functions myself.
I create a variable “MHI_AC_Ctrl_Core mhi_ac_ctrl_core;”.
In Setup() I call “ mhi_ac_ctrl_core.init();”...
In the following I wanted to call further mhi_ac_ctrl_core. ... functions.
I did not understand what the call “mhi_ac_ctrl_core.MHIAcCtrlStatus(&mhiStatusHandler);” in the setup of the original sketch is for...
Best regards
Haegar

glsf91, 2025-01-03:
The call  "mhi_ac_ctrl_core.MHIAcCtrlStatus(&mhiStatusHandler);" is taking care that cbiStatusFunction in MHI-AC-Ctrl.ino is called when in MHI-AC-Ctrl-core.cpp "m_cbiStatus->cbiStatusFunction....." is called.
Basically: the pointer to the class StatusHandler is passed through.

closed 2025-01-04 by Haegar88