absalom-muc/MHI-AC-Ctrl #205: Wemos restarts after connecting to AC

Haegar88, 2025-01-04:
Hi, after connecting to the AC the Flash-LED of the Wemos periodically flashes. I think he's rebooting periodically, but i dont't know why. Can I connect the Laptop to the Mirco-USB while the Wemos is powered via 12V from the AC? Any ideas why the device reboots?
BR Haegar

glsf91, 2025-01-04:
yes, you can connect your laptop if it is powered from ac.
Take a look at the serial logging to take a look if it is rebooting and why

Haegar88, 2025-01-04:
That's the serial output..

Starting MHI-AC-Ctrl v2.8
CPU frequency[Hz]=80000000
ESP.getCoreVersion()=3.1.2
ESP.getSdkVersion()=2.2.2-dev(38a443e)
ESP.checkFlashCRC()=1
Measure frequency for SCK, MOSI and MISO pin
SCK frequency=3200Hz (expected: >3000Hz) o.k.
MOSI frequency=615Hz (expected: <SCK frequency) o.k.
MISO frequency=0Hz (expected: ~0Hz) o.k.
OTA Ready
setupWiFi: Start async scanNetworks
WIFI_CONNECT_SCANNING
status=65 topic=Mode payload=Auto

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (3):
epc1=0x4021ba5e epc2=0x00000000 epc3=0x00000000 excvaddr=0x4024c938 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd80 end: 3fffffd0 offset: 0150
3ffffed0:  4024c938 0000010f 00000079 3ffeeb08  
3ffffee0:  4024c9e4 4024c938 00000041 402055e0  
3ffffef0:  2d49484d 432d4341 2f6c7274 65646f4d  
3fffff00:  00000000 00000000 00000000 00000000  
3fffff10:  00000000 00000000 00000000 00000000  
3fffff20:  00000000 00000000 00000000 00000000  
3fffff30:  00000000 00000000 00000000 00000000  
3fffff40:  00000000 00000000 00000000 00000000  
3fffff50:  00000000 00000001 0000000c 3ffeeb08  
3fffff60:  00000014 3ffeeac8 3ffe85c8 402013e4  
3fffff70:  00000053 00000000 00000001 000007fe  
3fffff80:  00000050 3ffe85db b3333333 00093e7f  
3fffff90:  3ffefa0c 40205c84 3ffeeb54 3ffef004  
3fffffa0:  00000000 3ffeeafc 3ffeeb08 40204e5f  
3fffffb0:  3fffdad0 00000000 3ffeefd8 40210478  
3fffffc0:  feefeffe feefeffe 3fffdab0 401003e5  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:1, boot mode:(3,3)

load 0x4010f000, len 3424, room 16 
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v000557e0
~ld

glsf91, 2025-01-04:
I don't know what you removed in the code (what was your plan).  Looks like when connecting to wifi it restarts. Use the esp exception decoder.

Haegar88, 2025-01-04:
Hello, the culprit was indeed Pubsub version 2.8.0 . With 2.7.0 it works without any problems. Thank you very much, I would never have thought of that.

closed 2025-01-05 by glsf91