absalom-muc/MHI-AC-Ctrl #207: Use in parallel with WF-RAC module?

ThinkPadNL, 2025-02-07:
Is it possible to use this module together/in parallel with the built-in WF-RAC module?

The Smart M-Air app which controls this WF-RAC is fine for me, but i would like to use an external room temperature for better temperature stability.

There is a ready for use product that can do this, but i find it too expensive and also is not able to create weekly schedules like the Smart M-Air app does: https://lectronz.com/products/climacontrol-mhi-air-to-air

glsf91, 2025-02-08:
No that is not possible. It also won't start if it detects another slave signal (wf-rac). Only 1 slave can send data to AC

ThinkPadNL, 2025-02-08:
How does the product work that i linked then? That proves that technically it should be possible, maybe that device sits as a “man-in-the-middle” to intercept data from the WF-RAC?

I guess I’ll stick to my original plan of extending the 5K NTC room thermistor then.

glsf91, 2025-02-08:
I think it is indeed a man in the middle.

fonske, 2025-02-08:
@ThinkPadNL 
Maybe you can ask if he makes his code public? To see how he does it?
As all the example code he might have used is also open source he needs to share his code.

glsf91, 2025-02-19:
I think we can close this one. Otherwise I can reopen

closed 2025-02-21 by ThinkPadNL

blues-sechseck, 2026-09-04:
Late to this, but for anyone who finds the thread later: there is a second
route, and it doesn't need the module removed.

The WF-RAC module's own local API carries a room temperature in the command
frame — `command[5]`, `raw = round(T × 4) + 61`, `0xFF` meaning "use the
internal sensor". Writing a value below `0xFF` makes the AC control on it. That
is the same thing `Troom` does here, on the same 0.25 K grid, just over the
module instead of the CNS bus. Verified on an SRK20ZS-WF; I wrote up the
details while working on the Home Assistant integration for the module:
https://github.com/blues-sechseck/Mitsubishi-WF-RAC-Integration/blob/main/docs/wf-rac-module-reference.md#56-injecting-an-external-room-temperature

Nothing above contradicts the answer in this thread: two slaves on the bus
really don't work, so running both at once is still out. This is the other side
of the same wall — no external sensor wiring, no thermistor extension, and the
app keeps working.

Two caveats, since they cost me time: the byte has no set-bit, so any frame
that leaves it at `0xFF` puts the unit back on its internal sensor — the value
has to be re-sent on every write. And while it stands, the unit reports the
injected value back, so its own reading isn't on the wire any more.