absalom-muc/MHI-AC-Ctrl #213: Idea: Introduce a passive mode to allow RC timer

git-developer, 2025-06-16:
A [known limitation](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/Troubleshooting.md#known-limitations) is that the timer of the IR remote control does no longer work as soon as a device is connected to the SPI port.

The README of project [hberntsen/mhi-ac-ctrl-esp32-c3](https://github.com/hberntsen/mhi-ac-ctrl-esp32-c3) contains a paragraph about a so called "active mode". If I understand it correctly, users can toggle between active and passive mode. Active mode is the current behavior of MHI-AC-Ctrl. In passive mode, the AC is monitored but cannot be controlled, and the RC timer functionality is working.

I tried to find [in the code](https://github.com/hberntsen/mhi-ac-ctrl-esp32-c3/blob/15c1237acc15840f4d1318576da16d1de2db1380/esphome/components/MhiAcCtrl/MHI-AC-Ctrl-core.cpp#L219) what happens behind the scenes but didn't understand how this feature works. I guess this might be some kind of disabling SPI so that the AC doesn't know that a device is connected.

Would such a passive mode be possible within MHI-AC-Ctrl, too? What do you think?

git-developer, 2025-06-18:
Thanks for joining! That sounds to me as if this feature could be implemented by introducing a new topic, e.g. `PassiveMode`, and sending `0xff` via SPI when enabled.

git-developer, 2025-07-06:
Thanks for joining! What do you mean by "disconnecting", physical removal of the wire or sending of a command?

absalom-muc, 2025-07-14:
physical removal

absalom-muc, 2025-11-29:
You implemented a passive/active mode switch during operation. Not sure if it can work. When I remember correctly discussions in the past, the active mode is entered once the AC sees a '0' on the MISO pin and you can't re-enter the passive mode for this power cycle. Or to say it differently: After powering (remove / reconnect power supply) the AC must not have a '0' on the MISO pin, a '1' on the MISO pin is required.
To confirm if my thinking is correct, I suggest you implement my [proposal](https://github.com/absalom-muc/MHI-AC-Ctrl/issues/213#issuecomment-3553748283), remove the power supply and re-connect the power supply. And then please try pressing one of the RC buttons TIMER ▲/▼

closed 2025-12-01 by absalom-muc